Staacks/inkkeys #1: Request: Gerber Files for Other PCB Services?

Extra-Fox, 2021-02-21:
Hi, I am totally digging this project and am eager to try my hand at building one. 

Being an absolute hack when it comes to PCB software, would it be possible for you to create some Gerber files of your PCB design? I tried plotting it from KiCAD, but I wasn't getting copper layers that JLCPCB recognized.

Staacks, 2021-02-23:
Hm... Not sure if it will help if I export them. I ordered my PCB from aisler.net where I uploaded the KiCad files, so I have no starting point to export them any differently than you did. Are there any naming conventions from JLCPCB? I am quite sure that I had to rename a layer or two to match the aisler naming conventions, so maybe it is as simple as naming the layers according to their suggestions.

Extra-Fox, 2021-02-23:
Okay, bear with me here because my skill set here barely exists.

But it looks like they were looking for two layers named *.cu. Which didn't appear when I tried plotting from KiCAD (I downloaded and installed it for this)

[image]

And I was following these instructions here: https://support.jlcpcb.com/article/44-how-to-export-kicad-pcb-to-gerber-files

So if it's simply a naming convention in that case, I'd be happy to go into the files I have locally and change that, however I'd probably need your guidance on the conversion key.

Staacks, 2021-02-23:
Ah, I see. The "F.Cu" and "B.Cu" are just named "Front" and "Back" in my file. And now that I checked it I realize that this was not my idea but I started with a template from Aisler.

You might want to compare the manufacturing tolerances of Aisler and JLCPCB to see if there is something you need to look out for. Since I used the Aisler rules, some distance could be too small for other services, although I would not expect it as there is nothing too fancy about this design.

Extra-Fox, 2021-02-23:
Okay, I'll do that, thank you for the heads up and the guidance. 

If anything, I can order from Aisler, but given my location it's generally less expensive for me to use JLCPCB. However, I do want this to work so maybe it's just safer to use Aisler.

closed 2021-02-23 by Extra-Fox

Staacks, 2021-02-23:
Yeah, I have no reason to push Aisler - it is just the one I new and happened to use here. This is also the first PCB I designed, but I have ordered from Aisler before in the context of a project from work (in this case the PCB was designed by my PhD student who has done a few PCBs before). To us they are the most local manufacturer (we could literally walk across the border to their headquarter), so I started there without comparing manufacturers a lot.

However, the thing that made me stick with Aisler at least for this first project was that they have a render guarantee. So, I just upload the KiCad file and their system renders a preview of the PCB and they guarantee that the manufactured PCB will match that preview. For me as a beginner this eliminated exactly the "are the gerber files correct?" uncertainty. I will probably try other sources as well on simpler future projects, because Aisler is indeed rather expensive.

Extra-Fox, 2021-02-23:
I totally understand and I didn't get that impression. Local is always good. We have OSHPark which might be considered local, but is still half way across the country from me. Their prices get rather high as well depending on the board.

I've never even designed a board before, just sent those designed by others off to be produced. So, I'm not sure I'm brave enough to take the leap you did, but maybe one day. So far I've just clumsily hand wired any projects without boards.

Extra-Fox, 2021-02-24:
Well, I figured it out. I got it working with OSHPark and with JLCPCB. Unfortunately OSHPark's price was over $70USD so that was a non-starter. JLCPCB was able to do it, with a black circuit board and ENIG-ROHS coating for $35USD shipped.

Staacks, 2021-02-24:
Well, that is not that much cheaper, but it is a rather large PCB.

Extra-Fox, 2021-02-24:
No, probably less than having it made in DE and shipped here, but not a lot less. I'm more familiar with using JLCPCB though so it's a comfort level thing I suppose.

Extra-Fox, 2021-03-16:
It sure is a nice board.
[image]

Staacks, 2021-03-19:
I really need to try other manufacturers - Just for that black finish :)

x4FF3, 2021-06-11:
my JLCPCB order was 16$ for 5 PCBs :) shipped to germany

WiBla, 2022-01-26:
Hello, I'm also looking forward building my own macropad !

@x4FF3 would you be so kind as to provide the gerber files you used ?
Thanks in advance !