Staacks/inkkeys #2: Capacitors for neo pixels

drobertwilcox, 2021-10-31:
Hello Sebastian,

I found your project through a hack a day blog post, https://hackaday.com/2021/02/18/dynamic-macro-keyboard-controls-all-the-things/, and I am certainly interested in trying to build one.  I have not completely reviewed the circuit, but I did see that the RGB neo pixel LED modules do not have a decoupling capacitor.  This could cause flashing, browning out, general misbehaving of the LED device.  I am currently installing ki-cad to look at the circuit design and layout.  I am bringing this up because I was thinking about modifying the layout and adding a string of addressable RGB under the key caps, because more RGB is always better! Though I will have to look at the power consumption of the devices in total to ensure that it stays USB compatible with no need of an external power source. This part may be the part of choice for individual key RGB https://kprepublic.com/products/xd-halo-rgb-light-control-pcb-soft-lighting-light-diffuser-circle-light-for-mechanical-keyboard-cospad-xd60-64-75-84-96-40-68

Staacks, 2021-10-31:
Yes, I definitely forgot those. I am quite new to these designs and learned about this issue after I finished the project. Luckily, I had no issues so far, but as far as I can tell from my research, the cheap THT RGB LEDs in my design are even more susceptible to such issues than the typical WS2812b ones.

Let me know about your project when it is ready. If you intent to share, I will happily link to the improved version in the readme of my repository version or directly add it if you don't want to publish it as a fork yoursef.

drobertwilcox, 2021-10-31:
I am new to GIThub, please excuse me if I don't know the processes that are in place to fork a project.  I will be exploring options for the RGB key switches.  Maybe the key switches with the 4 holes for the LED to pass through would be a better choice for anchoring the key to the board: https://www.amazon.com/Transparent-Switches-Mechanical-Keyboard-Compatible/dp/B07K7LJNLR?th=1 I don't know if the 5mm LED will fit inside.  
I will let you know what I come up with, and also I may need some time to learn how to use github.