Staacks/inkkeys #4: LED Pin Spacing and Ground Plane

Rudis1261, 2022-01-12:
I order 5 boards from PCBway and I am super happy that they came out really great. But one of the issues I was having is definitely the pin spacing of the LEDs, it's quite a pain. I think most people would be happy to print a 3mm or so ring spacer and rather use 2.54mm pitch, with the legs slightly bent. I kinda trashed one of my boards just because of that issue.

This raised something else, which may need improvement. The ground plane is massive :scream:, my poor little soldering iron could not heat it fast enough to use a wick as suggested to clear out excess solder.

I wonder if the ground plane could have an isolation ring around the circuit to reduce the ground plane, and make it easier to heat up. Basically, all the heat is transferring into that entire PCB plane at the moment.

Not so much as an issue, but a quality of life on both

Staacks, 2022-01-14:
I am happy to add this to the repository if someone tries this modification and confirms that it works.

Regarding the GND plane, I do not particularly recall problems while soldering, but for anyone who is worried about this, there is an easy solution: If you select the ground plane in KiCad and go to its properties, there is an option "pad connections" which is already set to "thermal reliefs". So, at the moment, all pads only use thin spokes to connect to the ground plane to alleviate this exact problem. If you want to reduce the thermal coupling further, you can modify the settings below this to create thinner spokes or increase the clearance. You can also set this individually for each pad than for the entire plane.

Rudis1261, 2022-01-14:
That's awesome thanks Sebastian! I will probably have it made in future
again as the project matures.

I will also definitely try your suggestion next time before I generate the
grbl files.

Cheers,
Rudi