Staacks/inkkeys #5: Can't seem to get the python app running

Rudis1261, 2022-01-13:
I run windows 10, and it seems it's not quite working as expected. I first ran into an issue where win32gui wasn't available, and resolved that by installing `pip install pywin32`

Then I tried launching it, but next it failed at a pulse audio import in modes.py. Which from what I understand should only be available in linux. I then commented out the pulse audio import which got me to where I am now. 

My device is COM20, and when I start it, it does this:
```$ ./controller.py 
https://there.oughta.be/a/macro-keyboard
I will try to stay connected. Press Ctrl+c to quit.
Connecting to  COM20 .
Requesting device info...
Header found. Waiting for infos...
End of info received.
Testmode:  False
Number of LEDs:  10
Display width:  128
Display height:  296
Rotation circle steps:  20
Connected to  COM20 .
Error:  <class 'AttributeError'>
I will retry in three seconds...
Connecting to  COM20 .
Serial error:  could not open port 'COM20': PermissionError(13, 'Access is denied.', None, 5)
I will retry in three seconds...
Connecting to  COM20 .
Serial error:  could not open port 'COM20': PermissionError(13, 'Access is denied.', None, 5)
I will retry in three seconds...
Connecting to  COM20 .
Serial error:  could not open port 'COM20': PermissionError(13, 'Access is denied.', None, 5)
I will retry in three seconds...
Connecting to  COM20 .
Serial error:  could not open port 'COM20': PermissionError(13, 'Access is denied.', None, 5)
I will retry in three seconds...
Connecting to  COM20 .```

Any advice / suggestions would be welcome.

Staacks, 2022-01-16:
Unfortunately, I do not have a Windows machine that I can easily use to test, so I can only offer my experience of a little mistake I sometimes made during development: If you have the Arduino IDE and especially the serial monitor open at the same time, the serial port is already in use and you cannot open it in Python.
If it's not that simple, I would suggest to try open COM20 with a minimal example from pyserial first to narrow down possible reasons.

Rudis1261, 2022-01-17:
I think it's this bit causing the issue: ```Error:  <class 'AttributeError'>``` it connects and gets the info, but after that, I think it's tied up by the process before the failure. 

I don't know python's debugger really, otherwise, I would step through the code and debug it. I am assuming it's because of the pulse library methods being called which I had to comment out to get it to even run in the first place. I will try and debug it at some point yeah.

Rudis1261, 2022-01-18:
Bricked my leo, can't continue, so closing the issue.

closed 2022-01-18 by Rudis1261

Staacks, 2022-01-18:
Again, sorry to hear that.

If you manage to get it running again or to anyone else stumbling upon this:
You are right, I missed the first error in the log. Unfortunately, that does not tell much. If you change line 4 of the Python script from "DEBUG = False" to "DEBUG = True", we should see a stack trace, which allows us to understand where this is comming from.

Rudis1261, 2022-01-18:
Will definitely try again once I get more stock of all the parts thanks

amercer86, 2022-01-22:
I think this COM error is a red herring. I was experiencing the exact same issues with my windows install as well. First I needed to disable pulsectl, then I started getting the COM Permission Errors.

I switched DEBUG to True and saw that MQTT was failing to connect (I hadn't tried to configure this yet). I updated the IP to connect to my broker - and the COM errors stopped. I switched back and forth to confirm.

It seems to me like MQTT connection failure is reporting as this COM error.