Staacks/inkkeys #6: Example Of Images

Rudis1261, 2022-01-15:
Is there any chance I could get an example of a full image refresh and a partial image refresh? 

I am a bit confused and trying to get something usable going in golang, but I need an example of how to flash the whole screen and a partial update. What it would look like in raw serial data would be very useful. 

Like a simple example of writing a centered glyph or word, and the steps to draw it :pray:

Staacks, 2022-01-16:
Can you run the Python controller? If so, the easiest way would probably be logging its serial communication, as it would make it quite easy for you to create specific examples that you are interested in.

Rudis1261, 2022-01-17:
It breaks on windows, so doesn't get to that point sadly.

Rudis1261, 2022-01-18:
I bricked my leo last night trying to do a test image with the native library to test. So closing the issue

closed 2022-01-18 by Rudis1261

Staacks, 2022-01-18:
Sorry to hear that. Can't you just reinstall a bootloader with another Microcontroller?

(Also: Why a Leonardo? Were you trying it on a breadboard?)

Rudis1261, 2022-01-18:
I believe the micro pro is just the modern version of the leonardo. Their pinout is the same and everything. At least when I plug one in, its what my arduino recognizes it as.

I got the display to work just before bricking it 🙈🙈 I had one of the images on memory and with a serial command it printed it to screen which finally worked. 

And then I accidentally uploaded too large a sketch which bricked it 🙈

Rudis1261, 2022-01-18:
I will try and reinstall the bootloader with an ardiuno as isp. But I doubt it will work, looks like its completely locked up. Oh well serves me well for being impatient

amercer86, 2022-01-22:
I'm also using a Leonardo - it _is_ labeled as a Micro Pro though