Staacks/inkkeys #7: inkkeys/python-controller/processchecks.py

s53zo, 2022-01-16:
Hi!
Thanks for making this great project. I will build 6 of the keyboards to help control a radio amateur station.

On a Mac OS, Monterey 12.1 I have a problem with using AppKit. It is installed with pip3 so the OS should be ready.

when processchecks.py is with capital letters like:
from AppKit import NSWorkspace

it doesn't find the AppKit. If I change it to small caps:
from appkit import NSWorkspace

it finds the Appkit but doesn't find NSWorkspace with error:
Traceback (most recent call last):
  File "/Users/simon/Downloads/inkkeys/python-controller/controller.py", line 7, in <module>
    from processchecks import *  #Functions to check for active processes and windows
  File "/Users/simon/Downloads/inkkeys/python-controller/processchecks.py", line 13, in <module>
    from appkit import NSWorkspace
ImportError: cannot import name 'NSWorkspace' from 'appkit' (/usr/local/lib/python3.9/site-packages/appkit/__init__.py)

Any idea and help please?

Staacks, 2022-01-16:
Unfortunately, I cannot try it myself on a Mac and this part of the code is from an example without actual testing (as mentioned in the article). However, the fact that for you it is lower-case is quite suspicious and it seems like you are trying to load NSWorkspace from a different package.
I found this: https://stackoverflow.com/questions/53808723/importerror-no-module-named-appkit-after-installing-appkit-and-pyobjc
Could this be your problem, too?

s53zo, 2022-01-17:
Good call, exactly as you foresaw.

pip3 uninstall AppKit 
pip3 install --upgrade --force-reinstall PyObjC PyObjC-core

solved the issue.

closed 2022-01-17 by s53zo