Staacks/inkkeys #11: Question regarding resistor cluster

Rudis1261, 2022-01-22:
[image]

I am curious what these resistors are used for, mind sharing some insight? I may or may not be busy with a variation of the board, and I can't say I have played with an encoder wheel before to understand why it's needed.

Rudis1261, 2022-01-22:
Are they serving as a voltage devider?

Staacks, 2022-01-22:
No, to my understanding, it is a pull-up and a low-pass for debouncing. I have to admit that I found this as one of the most common ways to hook up a rotary encoder and I did not much testing, so there might be details I am missing myself, but I think I have a good idea of what it does. First of all, we should look at the schematic for a more intuitive layout:
[image]
(I removed the irrelevant parts.)

You will notice that the encoder is only connected to ground and does not have any positive connection. S2 is the push button on the encoder and if you press it, it closes the connection to S1, which is connected to ground. In order to have a proper state while it is not pressed, it needs a pull-up resistor, which is no problem as we use the internal pull-up of the Pro Micro on pin 20.

For the rotation, the situation is similar. contacts A and B will alternate between being open or connected to C, which in turn is connected to ground. So, for the open state, we need a pull-up. In a perfect world, we could connect A and B directly to pins 1 and 2 on the Pro Micro and also use internal pull-ups. Unfortunately, any switch might bounce (quickly alternate between open and closed before eventually settling on a final state).

For the normal keys and the push button, this is not that much of a problem, because we just lazily read them in a loop and bouncing is usually faster than our read-out anyways. The rest can be done with a debounce in software to avoid multiple triggers within a certain time frame. If it wasn't a macro keyboard but something with very rapid inputs, this might not be enough, but for a macro keyboard, this should be fine.

On the rotary encoder however, we might want to quickly turn it for fast scrolling, so I use a fast encoder library which uses interrupts on both pins that are connected to the encoder (as I said, I never tested it without, so it might be fast enough without interrupts). If a bounce occurs now, the interrupts will be fast enough to trigger multiple times and as the relative switches of both inputs might even be identical to a full encoding cycle (in contrast to a key, where there are only two states), we really need to avoid this and it becomes really hard to detect this in software. So, we debounce in hardware.

To my understanding, C1 and R4 as well as C2 and R3 act as a simple low pass. Thinking about it now, intuitively I would have placed the capacitors to ground between the resistors and the GPIO pins, but I don't think that this makes much of a difference. 

Now, that we have placed those resistors between pin A and B and the GPIOs, using the internal pull-ups would be quite pointless. They would only permanently the GPIOs to +5V as the encoder would not be able to pull it back to ground through that 10k resistor. So, instead, we need to add external pull-ups before that, which is R1 and R2.

Hope, you could follow - and I hope that my own interpretation is correct :)

Rudis1261, 2022-01-23:
That is great thanks for the explanation, that helps a lot!  😃

closed 2022-01-23 by Rudis1261