Staacks/inkkeys #13: PCB give away if you cover my costs

Helicon007, 2022-01-30:
I ordered a amount of 10 PCBs because one piece is obviously not worth it. So I can share the remaining 9 PCBs if you cover my basics costs.

I'll send you one PCB each by postal mail without warranty it will arrive the destination. 2€ + postage fee. One mail is less than 40g.
Payment via Paypal.

Please contact me if you desire a brand new green PCB for the macro keyboard project.

btw: I have very good experience with JLCPCB from China - no advertisement but this was not my fist order and price/quality is very good.

Staacks, 2022-01-30:
Good idea to combine PCB orders, so I just shared this issue on Twitter. (Not sure if the issue tracker is the best place for it, though.)

ondras, 2022-01-30:
Hi @Helicon007,

I would be interested in the PCB - or perhaps two, if the postage does not increase dramatically. I am located in the Czech Republic. Is it okay to contact you this way? Can we arrange a shipment?

Helicon007, 2022-01-31:
> Hi @Helicon007,
> 
> I would be interested in the PCB - or perhaps two, if the postage does not increase dramatically. I am located in the Czech Republic. Is it okay to contact you this way? Can we arrange a shipment?

Hi Ondras,
I sent you an eMail to handle this, hope it's not in your spam.
Regards, Helicon

marengo560, 2022-02-01:
Hi @Helicon007,

I would also be interested in a couple of them if you still have them.

Regards,

Rafael

Helicon007, 2022-02-01:
> Hi @Helicon007,
> 
> I would also be interested in a couple of them if you still have them.
> 
> Regards,
> 
> Rafael

Sure, contact me via helicon@gmx.de
Let me know how many and what address, I‘ll check the postage then. 

Regards, Helicon

Helicon007, 2022-02-02:
The PCB's are on their way and I'm glad I didn't waste them. I just hope they receive their destination undamaged. If not, let me know please.

Have fun, Helicon

closed 2022-02-02 by Helicon007

marengo560, 2022-02-08:
Hi @Helicon007,

Just to let you know I got the PCBs and they arrived safe and sound! Thanks 👍 

Regards,

Rafael