Staacks/inkkeys #14: e-Paper Module not Performing Parital Refresh

lucas-fern, 2022-02-20:
I've got the device programmed and working on Windows 10, but haven't been able to get the display to perform a partial refresh at any point. I have the v2 of the display and have made the changes to the arduino code as specified in the readme.

Not sure what the intended behaviour is, but the hardware demo file does display a static image, and then clears.

In the python controller however the functionality of the different keys is working, but the only time I can get an image on the screen is by pressing button 8 to toggle the demo display. I realised that this function makes a call to `updateDisplay` with `fullRefresh=True`, so I made that the default in the `updateDisplay` function and now all the screens display correctly, but only after a lengthy full refresh.

Has anyone else encountered this issue? Or any suggestions for what to try to resolve it?

Staacks, 2022-02-21:
Sounds like this one discussed over at Reddit:
https://www.reddit.com/r/thereoughtabe/comments/s9sgox/inkkey_display_wont_update/

Is there something in there, that solves it for you?

lucas-fern, 2022-02-22:
Yep, amazing. Made all the changes suggested in that thread and partial updates are working now.

Cheers!

closed 2022-02-22 by lucas-fern