Staacks/inkkeys #15: Keycap model source?

ondras, 2022-03-11:
Hi,

I am trying to build this cool keypad. I got most of the stuff working okay, but I failed at keycaps: I managed to print them in PETG without issues, but they do not fit. The cross-shaped hole is too small (using typical Gateron switches).

I would like to fix this, but I need some source 3d model to work with. I do not have Blender nor KiCad (never used these). I am experienced with OpenSCAD and I suppose I can learn some other CAD tools as well, provided I can get the source model.

The documentation states that 3d models were taken from a KiCad component library. Can you please provide some more specific instructions/links for that?

Thanks,
Ondrej Zara

Staacks, 2022-03-13:
Sorry, the files in the repository is all I got. I design those models directly in Blender - not because it is a good tool for technical designs, but it is the tool I am familiar with. The downside is that it is difficult to do "non-destructive" modelling and at some point one often has to apply modifiers. I could not find a blend file with the original modifiers, so the mesh is all I have got (although it is pretty much just a boolean combination of the keycap, a cylinder and an extruded cross - it does not even have a concave top).

The 3d models from the KiCad component library are for the electronics. KiCad has a tool to import 3d models for most common components: https://kicad.github.io/packages3d/ - but that is probably not what you are looking for.

ondras, 2022-03-13:
No problem. I found https://www.thingiverse.com/thing:417509 in the meantime and I am 3d-printing eight of these right now. They come with OpenSCAD source, so I enlarged the cross-shaped hole and the fit is perfect now.

closed 2022-03-13 by ondras

Staacks, 2022-03-13:
Just a head-up: The spacing between the keys is a bit smaller than usual to fit four of them next to the display. Not sure if that's an issue with that round model.

ondras, 2022-03-13:
I had to shrink them. The spacing is an issue indeed: as a custom keyboard fan, I have quite a number of spare (regular) keycaps lying around. None of them fit, the PCB is too crowded. The only way is to 3d-print.

Staacks, 2022-03-13:
Jepp, unfortunately, the normal spacing spreads the keys beyond the length of the display. (But I also have to admit that I did not check the specs when designing the PCB.)