Staacks/inkkeys #18: GxEPD2 compiling error

Yersi88, 2022-10-16:
I keep getting this error when I try to complie the code. Any ideas how to fix thsi? 

```
In file included from c:\users\boss\documents\arduino\libraries\hid-project\src\hid-apis\KeyboardAPI.h:29:0,
                 from c:\users\boss\documents\arduino\libraries\hid-project\src\hid-apis\defaultkeyboardapi.h:27,
                 from C:\Users\BOSS\Documents\Arduino\libraries\HID-Project\src/SingleReport/BootKeyboard.h:30,
                 from C:\Users\BOSS\Documents\Arduino\libraries\HID-Project\src/HID-Project.h:50,
                 from C:\Users\BOSS\Downloads\inkkeys-main(1)\inkkeys-main\arduino\inkkeys\inkkeys.ino:13:
c:\users\boss\documents\arduino\libraries\hid-project\src\keyboardlayouts\improvedkeylayouts.h:54:21: note: #pragma message: Using default ASCII layout for keyboard modules
     #pragma message "Using default ASCII layout for keyboard modules"
                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from C:\Users\BOSS\Downloads\inkkeys-main(1)\inkkeys-main\arduino\inkkeys\inkkeys.ino:15:0:
C:\Users\BOSS\Documents\Arduino\libraries\GxEPD2\src/GxEPD2_BW.h:79:1: error: expected class-name before '{' token
 {
 ^
In file included from C:\Users\BOSS\Downloads\inkkeys-main(1)\inkkeys-main\arduino\inkkeys\inkkeys.ino:15:0:
C:\Users\BOSS\Documents\Arduino\libraries\GxEPD2\src/GxEPD2_BW.h: In constructor 'GxEPD2_BW<GxEPD2_Type, page_height>::GxEPD2_BW(GxEPD2_Type)':
C:\Users\BOSS\Documents\Arduino\libraries\GxEPD2\src/GxEPD2_BW.h:30:31: error: class 'GxEPD2_BW<GxEPD2_Type, page_height>' does not have any field named 'Adafruit_GFX'
 #define GxEPD2_GFX_BASE_CLASS Adafruit_GFX
                               ^
C:\Users\BOSS\Documents\Arduino\libraries\GxEPD2\src/GxEPD2_BW.h:85:44: note: in expansion of macro 'GxEPD2_GFX_BASE_CLASS'
     GxEPD2_BW(GxEPD2_Type epd2_instance) : GxEPD2_GFX_BASE_CLASS(GxEPD2_Type::WIDTH, GxEPD2_Type::HEIGHT), epd2(epd2_instance)
                                            ^~~~~~~~~~~~~~~~~~~~~
In file included from C:\Users\BOSS\Downloads\inkkeys-main(1)\inkkeys-main\arduino\inkkeys\inkkeys.ino:15:0:
C:\Users\BOSS\Documents\Arduino\libraries\GxEPD2\src/GxEPD2_BW.h:89:17: error: 'HEIGHT' was not declared in this scope
       _pages = (HEIGHT / _page_height) + ((HEIGHT % _page_height) > 0);
                 ^~~~~~
C:\Users\BOSS\Documents\Arduino\libraries\GxEPD2\src/GxEPD2_BW.h:89:17: note: suggested alternative: 'HIGH'
       _pages = (HEIGHT / _page_height) + ((HEIGHT % _page_height) > 0);
                 ^~~~~~
                 HIGH
C:\Users\BOSS\Documents\Arduino\libraries\GxEPD2\src/GxEPD2_BW.h: In member function 'void GxEPD2_BW<GxEPD2_Type, page_height>::drawPixel(int16_t, int16_t, uint16_t)':
C:\Users\BOSS\Documents\Arduino\libraries\GxEPD2\src/GxEPD2_BW.h:122:15: error: 'WIDTH' was not declared in this scope
           x = WIDTH - x - 1;
               ^~~~~
C:\Users\BOSS\Documents\Arduino\libraries\GxEPD2\src/GxEPD2_BW.h:126:15: error: 'HEIGHT' was not declared in this scope
           y = HEIGHT - y - 1;
               ^~~~~~
C:\Users\BOSS\Documents\Arduino\libraries\GxEPD2\src/GxEPD2_BW.h:126:15: note: suggested alternative: 'HIGH'
           y = HEIGHT - y - 1;
               ^~~~~~
               HIGH
C:\Users\BOSS\Documents\Arduino\libraries\GxEPD2\src/GxEPD2_BW.h: In member function 'void GxEPD2_BW<GxEPD2_Type, page_height>::display(bool)':
C:\Users\BOSS\Documents\Arduino\libraries\GxEPD2\src/GxEPD2_BW.h:196:63: error: 'WIDTH' was not declared in this scope
       if (partial_update_mode) epd2.writeImage(_buffer, 0, 0, WIDTH, _page_height);
                                                               ^~~~~
C:\Users\BOSS\Documents\Arduino\libraries\GxEPD2\src/GxEPD2_BW.h:197:57: error: 'WIDTH' was not declared in this scope
       else epd2.writeImageForFullRefresh(_buffer, 0, 0, WIDTH, _page_height);
                                                         ^~~~~
C:\Users\BOSS\Documents\Arduino\libraries\GxEPD2\src/GxEPD2_BW.h:201:45: error: 'WIDTH' was not declared in this scope
         epd2.writeImageAgain(_buffer, 0, 0, WIDTH, _page_height);
                                             ^~~~~
C:\Users\BOSS\Documents\Arduino\libraries\GxEPD2\src/GxEPD2_BW.h: In member function 'void GxEPD2_BW<GxEPD2_Type, page_height>::displayWindow(uint16_t, uint16_t, uint16_t, uint16_t)':
C:\Users\BOSS\Documents\Arduino\libraries\GxEPD2\src/GxEPD2_BW.h:219:36: error: 'HEIGHT' was not declared in this scope
       uint16_t y_part = _reverse ? HEIGHT - h - y : y;
                                    ^~~~~~
C:\Users\BOSS\Documents\Arduino\libraries\GxEPD2\src/GxEPD2_BW.h:219:36: note: suggested alternative: 'HIGH'
       uint16_t y_part = _reverse ? HEIGHT - h - y : y;
                                    ^~~~~~
                                    HIGH
C:\Users\BOSS\Documents\Arduino\libraries\GxEPD2\src/GxEPD2_BW.h:220:47: error: 'WIDTH' was not declared in this scope
       epd2.writeImagePart(_buffer, x, y_part, WIDTH, _page_height, x, y, w, h);
                                               ^~~~~
C:\Users\BOSS\Documents\Arduino\libraries\GxEPD2\src/GxEPD2_BW.h: In member function 'void GxEPD2_BW<GxEPD2_Type, page_height>::setFullWindow()':
C:\Users\BOSS\Documents\Arduino\libraries\GxEPD2\src/GxEPD2_BW.h:233:15: error: 'WIDTH' was not declared in this scope
       _pw_w = WIDTH;
               ^~~~~
C:\Users\BOSS\Documents\Arduino\libraries\GxEPD2\src/GxEPD2_BW.h:234:15: error: 'HEIGHT' was not declared in this scope
       _pw_h = HEIGHT;
               ^~~~~~
[... 85 lines cut ...]
             epd2.writeImageAgain(_buffer, 0, page_ys, WIDTH, gx_uint16_min(_page_height, HEIGHT - page_ys));
                                                       ^~~~~
C:\Users\BOSS\Documents\Arduino\libraries\GxEPD2\src/GxEPD2_BW.h:432:90: error: 'HEIGHT' was not declared in this scope
             epd2.writeImageAgain(_buffer, 0, page_ys, WIDTH, gx_uint16_min(_page_height, HEIGHT - page_ys));
                                                                                          ^~~~~~
C:\Users\BOSS\Documents\Arduino\libraries\GxEPD2\src/GxEPD2_BW.h:432:90: note: suggested alternative: 'HIGH'
             epd2.writeImageAgain(_buffer, 0, page_ys, WIDTH, gx_uint16_min(_page_height, HEIGHT - page_ys));
                                                                                          ^~~~~~
                                                                                          HIGH
C:\Users\BOSS\Documents\Arduino\libraries\GxEPD2\src/GxEPD2_BW.h: In member function 'void GxEPD2_BW<GxEPD2_Type, page_height>::_rotate(uint16_t&, uint16_t&, uint16_t&, uint16_t&)':
C:\Users\BOSS\Documents\Arduino\libraries\GxEPD2\src/GxEPD2_BW.h:584:15: error: 'WIDTH' was not declared in this scope
           x = WIDTH - x - w;
               ^~~~~
C:\Users\BOSS\Documents\Arduino\libraries\GxEPD2\src/GxEPD2_BW.h:588:15: error: 'HEIGHT' was not declared in this scope
           y = HEIGHT - y - h;
               ^~~~~~
C:\Users\BOSS\Documents\Arduino\libraries\GxEPD2\src/GxEPD2_BW.h:588:15: note: suggested alternative: 'HIGH'
           y = HEIGHT - y - h;
               ^~~~~~
               HIGH

exit status 1

Compilation error: exit status 1
```

Staacks, 2022-10-20:
Sorry, I did not yet have the time to check if this is an issue with the current version of the display library. Just as a first thought, so I can at least give some input: Did you try an older version of GxEPD2 from around the time I released my project? Also: Did you install the library manually or from the Arduino library manager? (So I can try it myself when I find some time.)

Yersi88, 2022-11-05:
Turns out this did not compile on windows, I compiled it without issues on arch!

Staacks, 2022-11-06:
Thanks for the feedback and sorry that I did not check it. Since my only Windows machine is purely used as a drum computer, I cannot really say anything about this problem, but I will leave the issue here in the hope that someone else can shed some light onto this problem. In my experience Arduino libraries are usually not too picky about how they are included and on which OS they are compiled (after all it's a cross-compiling process in any case).