Staacks/inkkeys #19: OSError: cannot open resource

Yersi88, 2022-11-21:
**I get this error when I try to run the python script:** 
```
Connecting to  /dev/ttyACM2 .
Requesting device info...
Sending: I
Received: Inkkeys
Header found. Waiting for infos...
Received: TEST 0
Received: N_LED 12
Received: DISP_W 128
Received: DISP_H 296
Received: ROT_CIRCLE_STEPS 20
Received: Done
End of info received.
Testmode:  False
Number of LEDs:  12
Display width:  128
Display height:  296
Rotation circle steps:  20
Connected to  /dev/ttyACM2 .
Active window: Navigator
Traceback (most recent call last):
  File "/home/marius/Downloads/inkkeys-main/python-controller/controller.py", line 110, in tryUsingPort
    work()  #Success, enter main loop
  File "/home/marius/Downloads/inkkeys-main/python-controller/controller.py", line 78, in work
    mode.activate(device)       # ...and call its activate function
  File "/home/marius/Downloads/inkkeys-main/python-controller/modes.py", line 206, in activate
    device.sendTextFor("title", "Default", inverted=True) #Title
  File "/home/marius/Downloads/inkkeys-main/python-controller/inkkeys/device.py", line 207, in sendTextFor
    font1 = ImageFont.truetype("font/Munro.ttf", 10)
  File "/usr/lib/python3.10/site-packages/PIL/ImageFont.py", line 976, in truetype
    return freetype(font)
  File "/usr/lib/python3.10/site-packages/PIL/ImageFont.py", line 973, in freetype
    return FreeTypeFont(font, size, index, encoding, layout_engine)
  File "/usr/lib/python3.10/site-packages/PIL/ImageFont.py", line 249, in __init__
    self.font = core.getfont(
OSError: cannot open resource

Error:  <class 'OSError'>
I will retry in three seconds...
```

**The error changes when terminal is in focus:** 
```
Connecting to  /dev/ttyACM2 .
Requesting device info...
Sending: I
Received: Inkkeys
Header found. Waiting for infos...
Received: TEST 0
Received: N_LED 12
Received: DISP_W 128
Received: DISP_H 296
Received: ROT_CIRCLE_STEPS 20
Received: Done
End of info received.
Testmode:  False
Number of LEDs:  12
Display width:  128
Display height:  296
Rotation circle steps:  20
Connected to  /dev/ttyACM2 .
Could not get active window:  <class 'Xlib.error.BadWindow'>
Traceback (most recent call last):
  File "/home/marius/Downloads/inkkeys-main/python-controller/controller.py", line 110, in tryUsingPort
    work()  #Success, enter main loop
  File "/home/marius/Downloads/inkkeys-main/python-controller/controller.py", line 72, in work
    if ("process" in i and i["process"] in processes) or ("activeWindow" in i and i["activeWindow"].match(activeWindow)) or not ("process" in i or "activeWindow" in i):
UnboundLocalError: local variable 'activeWindow' referenced before assignment

Error:  <class 'UnboundLocalError'>
I will retry in three seconds...
```

I had it working few times so I'm not sure what might be causing this error, it seemingly appears out of nowhere. Any idea what might be causing this?

Staacks, 2022-11-21:
Looks like the font cannot be loaded. Simple first thought: Could it be that you are launching the script from a different directory? The fonts are given as a relative path, so it breaks if you call the Python script from somewhere else.

Yersi88, 2022-11-22:
Oh yes, that was indeed the case. Thank you!

closed 2022-11-22 by Yersi88