Staacks/inkkeys #20: Spare PCBs for sale (cheap)

Yersi88, 2022-11-22:
Hi,  

I have 4 more PCBs here that I don't need, they are quite pretty with black solder mask and white silkscreen. Price £4 + postage (should be cheap, fits in letter)

Yersi88, 2023-03-11:
> Hello, how much would it cost to ship to US

£4

dpoulson, 2024-01-17:
Don't suppose you have any left? Would save me getting some more made (I'm in the UK).

If not, might be worth closing this issue so you don't get hassled. :)

Helicon007, 2024-01-20:
I do have one left if you still need one. It'f for free, just cover the post stamp costs Germany to UK.

dpoulson, 2024-01-20:
Fantastic. Sure, let me know how much and your paypal address and I'll send some money over. 

Should be able to see my email on my profile, rather than putting it in a random issue. ;)