Staacks/inkkeys #21: Serial error - Permission denied when accessing COM port

MrEggy, 2024-03-16:
I am trying to get the Python controller running on a Windows machine.  when I run controller.py it appears to connect to the keypad successfully.  It retrieves the info from the keypad.  However when it enters the main loop I get the following error:

Error:  <class 'AttributeError'>
I will retry in three seconds...
Connecting to COM6 .
Serial error:  could not open port 'COM6': PermissionError(13, 'Access is denied.', None, 5)

If I enable debug mode I get the following traceback:

Traceback (most recent call last):
  File "D:\Downloads\DIY\Macro Keypad\inkkeys-main\python-controller\controller.py", line 110, in tryUsingPort
    work()  #Success, enter main loop
    ^^^^^^
  File "D:\Downloads\DIY\Macro Keypad\inkkeys-main\python-controller\controller.py", line 85, in work
    pollInterval = mode.poll(device)
                   ^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'poll'

Any assistance that you can offer would be appreciated.

MrEggy, 2024-03-18:
I believe this access denied error may in fact be spurious and actually caused by another error as indicated in Issue #5.  I believe it is caused by mqtt connection failing.  Then the serial connection subsequently fails.  If I disable the mqtt code the script runs for a little while.  It sends some keyboard mapping commands.  Which are successfully received by the keyboard.  Then the script exits with no exception.  Any advice on how to get this working on Windows would be greatly appreciated.  Likewise if anyone has some working code I would be very appreciative.

Staacks, 2024-03-19:
I am afraid that I will not be of much help with Windows-specific problems and especially how to set up Python on Windows (in my Linux world, installing Python is a one-liner).
However, if mode is None I would guess that you have entirely removed the fallback mode from the modes list. So, when none of the other modes feels "responsible" it will not default to the ModeFallback in the list. I have not looked into my own code for a while, but I think you need to remove the last parameter of the last entry in the "modes" list so it will always be used if the previous modes do not match.

MrEggy, 2024-03-20:
Thanks Sebastian.  I have just got to the bottom of this problem.  The permission error on the COM port is indeed spurious.  Whenever another error occurs it will generate this error.  Usually the root cause of the problem can be found by enabling debug mode.
Whilst I probably should have opened another issue for this.  The other issue that I reported where the script appears to run.  It communicates with the keyboard.  But then silently exits is caused by the following.
It appears that the getsize() method has been removed from Pillow in versions 10.0.0 or greater.  This will generate the following error.

AttributeError: 'FreeTypeFont' object has no attribute 'getsize'

The suggested solution is to downgrade to Pillow 9.5.0.  However it appears that Pillow 9.5.0 has a bug where it will just silently exit when this method is called.  I tried downgrading further however anything less than 9.5.0 is not compatible with Python 3.12.  So I was stuck.
The solution is to install Pillow version >= 10.0.0  As getsize() is missing you will need to replace the following lines in device.py.  

Replace line 208:
wt1, ht1 = font1.getsize(text);

with:
x1, y1, wt1, ht1 = font1.getbbox(text);

Replace line 210:
wt2, ht2 = font2.getsize_multiline(subtext);

with:
 x2, y2, wt2, ht2 = font2.getbbox(subtext);

Staacks, 2024-03-20:
Reopening as it is not yet fixed in the repo - as a reminder for me to fix it and so others who might run into it can find it easily.